Molorius/ADS126X #1: code does not compile

abchin, 2018-04-18:
Well, it does compile, but the problem seems to be in the linker step when compiling...

Linking everything together...
"C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.10.0_x86__mdqgnx93n4wtt\hardware\tools\avr/bin/avr-gcc" -w -Os -g -flto -fuse-linker-plugin -Wl,--gc-sections,--relax -mmcu=atmega2560  -o "C:\Users\ChinA\AppData\Local\Temp\arduino_build_121179/ADS1263.ino.elf" "C:\Users\ChinA\AppData\Local\Temp\arduino_build_121179\sketch\ADS1263.ino.cpp.o" "C:\Users\ChinA\AppData\Local\Temp\arduino_build_121179\sketch\ADS126X.cpp.o" "C:\Users\ChinA\AppData\Local\Temp\arduino_build_121179/core\core.a" "-LC:\Users\ChinA\AppData\Local\Temp\arduino_build_121179" -lm
C:\Users\ChinA\AppData\Local\Temp\ccQBJmNW.ltrans0.ltrans.o: In function `ADS126X::writeRegisters(unsigned char, unsigned char) [clone .constprop.23]':

ccQBJmNW.ltrans0.o:(.text+0x2fa): undefined reference to `_ads126x_write_pin_low(unsigned char)'

ccQBJmNW.ltrans0.o:(.text+0x326): undefined reference to `_ads126x_spi_rw(unsigned char*, unsigned char)'

ccQBJmNW.ltrans0.o:(.text+0x336): undefined reference to `_ads126x_write_pin_high(unsigned char)'

C:\Users\ChinA\AppData\Local\Temp\ccQBJmNW.ltrans0.ltrans.o: In function `ADS126X::sendCommand(unsigned char) [clone .constprop.18]':

ccQBJmNW.ltrans0.o:(.text+0x368): undefined reference to `_ads126x_write_pin_low(unsigned char)'

ccQBJmNW.ltrans0.o:(.text+0x374): undefined reference to `_ads126x_spi_rw(unsigned char*, unsigned char)'

ccQBJmNW.ltrans0.o:(.text+0x384): undefined reference to `_ads126x_write_pin_high(unsigned char)'

C:\Users\ChinA\AppData\Local\Temp\ccQBJmNW.ltrans0.ltrans.o: In function `main':

ccQBJmNW.ltrans0.o:(.text.startup+0x1d0): undefined reference to `_ads126x_setup_output(unsigned char)'

ccQBJmNW.ltrans0.o:(.text.startup+0x1d8): undefined reference to `_ads126x_write_pin_high(unsigned char)'

ccQBJmNW.ltrans0.o:(.text.startup+0x1dc): undefined reference to `_ads126x_spi_setup()'

ccQBJmNW.ltrans0.o:(.text.startup+0x1f0): undefined reference to `_ads126x_write_pin_low(unsigned char)'

ccQBJmNW.ltrans0.o:(.text.startup+0x20c): undefined reference to `_ads126x_spi_rw(unsigned char*, unsigned char)'

ccQBJmNW.ltrans0.o:(.text.startup+0x23c): undefined reference to `_ads126x_write_pin_high(unsigned char)'

ccQBJmNW.ltrans0.o:(.text.startup+0x24c): undefined reference to `_ads126x_write_pin_low(unsigned char)'

ccQBJmNW.ltrans0.o:(.text.startup+0x254): undefined reference to `_ads126x_delay(unsigned int)'

ccQBJmNW.ltrans0.o:(.text.startup+0x25c): undefined reference to `_ads126x_write_pin_high(unsigned char)'

ccQBJmNW.ltrans0.o:(.text.startup+0x2b2): undefined reference to `_ads126x_write_pin_low(unsigned char)'

ccQBJmNW.ltrans0.o:(.text.startup+0x2fa): undefined reference to `_ads126x_spi_rw(unsigned char*, unsigned char)'

ccQBJmNW.ltrans0.o:(.text.startup+0x436): undefined reference to `_ads126x_write_pin_high(unsigned char)'

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

abchin, 2018-04-18:
Found the issue....arduino.cpp....just renamed it to ADS126X_hardware.cpp

Molorius, 2018-04-18:
The example code compiles just fine for me on Linux with Arduino 1.8.5 for the Arduino Mega. I will test it on Windows 10 later, thanks for the tip.

gfs7x, 2018-05-17:
typedef union { // page 90
		struct {
			uint8_t CRC:2;            /*!< bit:  0.. 1 Checksum Byte Enable               */
			uint8_t STATUS:1;         /*!< bit:      2 Status Byte Enable                 */
			uint8_t TIMEOUT:1;        /*!< bit:      3 Serial Interface Time-Out Enable   */
			uint8_t :4;               /*!< bit:  4.. 7 Reserved                           */
		} bit;
		uint8_t reg;
	} ADS126X_INTERFACE_Type;

This piece of code is giving me troubles in Atmel studio. I really cannot figure out why. 

The errors is expected ')' before '*' token

Molorius, 2018-05-17:
Please paste the full error code. Also, you can put code in its own box by placing three backticks around it (```, see the [Markdown Cheatsheat](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet#code)). For example, this is that particular section of code:

```C++
typedef union { // page 90
  struct {
    uint8_t CRC:2;            /*!< bit:  0.. 1 Checksum Byte Enable               */
    uint8_t STATUS:1;         /*!< bit:      2 Status Byte Enable                 */
    uint8_t TIMEOUT:1;        /*!< bit:      3 Serial Interface Time-Out Enable   */
    uint8_t :4;               /*!< bit:  4.. 7 Reserved                           */
  } bit;
  uint8_t reg;
} ADS126X_INTERFACE_Type;
```

gfs7x, 2018-05-17:
The rest of the register mapping work fine, except for the interface type. Which indicates that union and struct can be combined to map registers. However, why atmel does give the error. I have no clue.

Molorius, 2018-05-17:
I cannot help with the issue unless you paste the full error code, similar to what you did in your first comment.

gfs7x, 2018-05-17:
`------ Build started: Project: datalogger, Configuration: Debug AVR ------
Build started.
Project "datalogger.cproj" (default targets):
Target "PreBuildEvent" skipped, due to false condition; ('$(PreBuildEvent)'!='') was evaluated as (''!='').
Target "CoreBuild" in file "C:\Program Files (x86)\Atmel\Studio\7.0\Vs\Compiler.targets" from project "C:\Users\gfs\Documents\Atmel Studio\7.0\work\datalogger\datalogger.cproj" (target "Build" depends on it):
	Task "RunCompilerTask"
		Shell Utils Path C:\Program Files (x86)\Atmel\Studio\7.0\shellUtils
		C:\Program Files (x86)\Atmel\Studio\7.0\shellUtils\make.exe all --jobs 4 --output-sync 
		Building file: .././main.c
		Invoking: AVR/GNU C Compiler : 5.4.0
		In file included from c:\program files (x86)\atmel\studio\7.0\toolchain\avr8\avr8-gnu-toolchain\avr\include\avr\io.h:550:0,
		                 from .././main.c:10:
C:\Users\gfs\Documents\Atmel Studio\7.0\work\datalogger\ADS126X_definitions.h(270,12): error: expected ')' before '*' token
		    uint8_t CRC:2;            /*!< bit:  0.. 1 Checksum Byte Enable               */
		            ^
C:\Users\gfs\Documents\Atmel Studio\7.0\work\datalogger\ADS126X_definitions.h(270,12): error: expected ')' before numeric constant
		    uint8_t CRC:2;            /*!< bit:  0.. 1 Checksum Byte Enable               */
		            ^
		make: *** [main.o] Error 1
		"C:\Program Files (x86)\Atmel\Studio\7.0\toolchain\avr8\avr8-gnu-toolchain\bin\avr-gcc.exe"  -x c -funsigned-char -funsigned-bitfields -DDEBUG  -I"C:\Program Files (x86)\Atmel\Studio\7.0\Packs\atmel\XMEGAA_DFP\1.1.68\include"  -O1 -ffunction-sections -fdata-sections -fpack-struct -fshort-enums -g2 -Wall -mmcu=atxmega256a3u -B "C:\Program Files (x86)\Atmel\Studio\7.0\Packs\atmel\XMEGAA_DFP\1.1.68\gcc\dev\atxmega256a3u" -c -std=gnu99 -MD -MP -MF "main.d" -MT"main.d" -MT"main.o"   -o "main.o" ".././main.c" 
C:\Users\gfs\Documents\Atmel Studio\7.0\work\datalogger\Debug\Makefile(86,1): error: recipe for target 'main.o' failed
	Done executing task "RunCompilerTask" -- FAILED.
Done building target "CoreBuild" in project "datalogger.cproj" -- FAILED.
Done building project "datalogger.cproj" -- FAILED.

Build FAILED.
========== Build: 0 succeeded or up-to-date, 1 failed, 0 skipped ==========`
Atmel error

Molorius, 2018-05-17:
I'm pretty sure the issue is that there is already a `CRC` constant defined for the atxmega256a3u microcontroller in avr-gcc, which would mess up the structure in ADS126X_definitions.h. The source of avr-gcc is not released, but I assume this because CRC is listed in the register map on page 64 of the [atxmega256a3u datasheet](http://ww1.microchip.com/downloads/en/DeviceDoc/Atmel-8386-8-and-16-bit-AVR-Microcontroller-ATxmega64A3U-128A3U-192A3U-256A3U_datasheet.pdf). Atmel tends to make their constants match those in the datasheet. 

You can test this by putting:
```C
#ifdef CRC
#error
#endif
```
before you include any of the ads126x headers in your main.c. If it throws an error, we've found your problem. 

You can fix this by changing every instance of `CRC` in my code to something else.

gfs7x, 2018-05-17:
Thank you for the response, yes that was the problem
`#ifdef CRC
#undef CRC
#endif`
This has solve the problem.

Molorius, 2018-05-17:
Fantastic, can this issue be closed then?

closed 2018-06-13 by Molorius