Molorius/ADS126X #4: IDAC1, IDAC2 and getting VREF from AIN0 to AIN5

q-bird, 2018-11-27:
Hi there,

1. Are there any functions to set IDAC1 and IDAC2 with the value from 50uA-3000uA as mentioned in the datasheet? IDAC is useful to measure RTDs or NTC...
2. With RTD sensor we need to read the REF voltage from AIN0-AIN1 or AIN2-AIN3 or AIN4-AIN5, not the internal ref, so is there any function to do it?
3. Is there any function to set ADC pin as single-ended input instead of diff-pair? And we also have AINCOM pin which can be the 11st ADC pin, how to measure from this pin?

Thank you!

Molorius, 2018-11-27:
1. This has not been implemented yet, but should not take very long. 
2. There is a pull request for this, see #3. 
3. You can set the negative side to ADS126X_ANALOG for the negative analog power supply monitor or ADS126X_DIGITAL for the negative digital power supply monitor to get single-ended input, depending on your use-case. AINCOM pin is set by ADS126X_AINCOM. All of this is documented in [the README](https://github.com/Molorius/ADS126X#int32_t-readadc1uint8_t-pos_pinuint8_t-neg_pin).

q-bird, 2019-01-26:
Hi @Molorius 
Any updates for IDAC?

Molorius, 2019-01-26:
IDAC functionality was added in 0be0e1f, but I don't have a solid way of testing it. Could you test to make sure it was implemented correctly?

q-bird, 2019-01-27:
@Molorius Thank you I will test it and get back to you soon

Molorius, 2019-02-26:
Fantastic, I will close this then.

closed 2019-02-26 by Molorius