Molorius/ADS126X #6: Wiring to arduino

Guerillah, 2018-12-04:
Is there a guide on how to wire up ADS1262 to the arduino (i.e. what this library expects PWDN/AVDD/DRDY/MISO/MOSI/etc.. to go to)?

ziatajo, 2018-12-08:
Hi Guerillah,
It depends on your Arduino model you are using. 
To run the basic example of this library, I think you could ignore the DRDY, START,PWDN pins (if you need them just pick a free GPIO Pin on your Arduino).
Connect the MISO, MOSI, SCLK of the ADS1262 to the MOSI, MISO, SCK on your Arduino.
CS pin to a free GPIO (and define it in the example)
AVDD --> 5V
AGND/DGND --> GND
DVDD --> 3V or 5V depending on the logic voltage your Arduino needs.

Hope that helps?

Molorius, 2018-12-11:
The SPI pins (MISO, MOSI, SCK) pins for the current Arduino lineup can be found [here](https://www.arduino.cc/en/Reference/SPI). The CS pin can go to the hardware CS pin, but I included an optional software CS pin so you can use the bus with multiple devices if needed (```begin()``` vs ```begin(uint8_t chip_select)```).

Guerillah, 2018-12-11:
I got it working by following the standard SPI pinout for Arduino Uno, i should have realized that by the library including <SPI.h>. Whoops!

Molorius, 2018-12-12:
Glad you got it working, I'm closing this issue.

closed 2018-12-12 by Molorius