Molorius/ADS126X #7: SPI clock frequency

AlbiBone, 2019-01-04:
Hi,

1)  I would like to increase the SPI clock frequency: I notice that ADS1262 allow a maximum frequency of 8 MHz, but I do not find the settings in the "Table of Contents".

2) In addition I notice that there is not #include "SPI.h" at the begin of your "ADS126X" library. Why?
How the ADS1262 comunicate with my Arduino?

Molorius, 2019-01-26:
1. This is currently hardcoded into src/boards/arduino.cpp. 
2. SPI.h is included in src/boards/arduino.cpp. 

You should be able to change the frequency, thanks for pointing that out. I will try to make a fix for this soon.

AlbiBone, 2019-01-26:
Thanks you for the info @Molorius , I will have a look.