Molorius/ADS126X #9: Multi-channel measurement ADS126x chip

AlbiBone, 2019-01-25:
Hi everyone,
since I would like to measure more than one analog signal, I am afraid about the synchronization of the analog channels.
Are the signals acquired simultaneously? If not, how can I overcome the problem of the voltages measured in times sligtly different for each channel?

Regards,
A

Molorius, 2019-01-26:
Are you positive you NEED simultaneous measurements? Both the ADS1262 and ADS1263 can take up to 38400 samples per second, which should be sufficient for most applications. The ADS1263 also has a 24 bit ADC, but this is intended for "background measurements". 

If you absolutely need simultaneous measurements, you may need to use a different ADC. I found many options by searching for "adc simultaneous sampling" online.

AlbiBone, 2019-01-26:
My question is referred due to the fact that if the to output node of the loadcell are not measured at the same time, the measurment is sliglty affected by errors (only on case of high frequency signals of force).
Anyhow, as you suggested @Molorius , using a high data rate (e.g. 38,4 kHz) can help to reduce this effect.
The last question: is it possible to determine the time lag between the to signals?

Molorius, 2022-05-28:
I am not positive, a careful reading of the datasheet is necessary to determine that.

closed 2022-05-28 by Molorius