Molorius/ADS126X #17: Shift input signal to 0-5V

cotixtof, 2022-06-08:
Hi Blake,
We continue to use your lib, thanks for your work.
When using the function disableInternalReference(), we have an error message when compiling (''undefined reference'') : how can we fix it?
Also, it seems we have an issue when setting the bias polarity(PULLUP mode), when AIN0 and AIN1 are floating we didn't read the AVSS. Thoughts?
Also when using AIN0 (positive) and AIN1 (negative) input, AVDD=DVDD and AVSS=DGND, we read +2500mV when AIN0 is floating and AIN1 is connected to AVSS, and we have -2500mV when AIN0 is floating and AIN1 is connected to AVSS, when we connect AIN0 to the positive side of a 1,5V battery and AIN1 to the negative side, we read 1500mV : thoughts?
Chris

Molorius, 2023-05-30:
1. disableInternalReference? I evidently never implemented this function. You will need to implement it in https://github.com/Molorius/ADS126X/blob/master/src/ADS126X.cpp.
2. Floating bias polarity? I would double check the datasheet on bias polarity. But overall I wouldn't trust floating pins on an ADC.
3. Partially floating pins? Again, double check the datasheet but be very wary of floating pins. I would not trust floating pins to give any useful data.

Molorius, 2023-05-30:
Note that I am not an electrical engineer though.