Molorius/ADS126X #18: Calibration functions wrong?

drsurfer, 2023-05-30:
The code for the calibration functions is very different from the "Calibration Command Procedure" listed in the ADS126X data sheet.
I doubt it works.
[image]

Molorius, 2023-05-30:
Thank you for pointing this out.

Note that this section was a recent addition to the datasheet. [This datasheet hosted by Mouser](https://www.mouser.com/pdfDocs/ads1260.pdf) was (at the time of this message) revised January 2019 and does not contain section 9.4.9. I don't remember what revision I used when writing this library. This is still an issue, but was not documented when this library was written.

drsurfer, 2023-05-31:
The datasheet you referred to is for the ADS1260/1 and - other than that it IS a different chip and has different calibration commands - it is a similar procedure documented in 9.4.7.5

The datasheet link for ADS1262 on Mouser point to current release on TI website. Googling for it you can find a Mouser hosted version that July 2015 revision, and even this _ancient_ release has the 9.4.9.8 paragraph. 

I think you simply missed that paragraph, after reading the previous ones, where are explained the various calibrations commands in an ambiguous manner. It's a kind of error I often do, too, when I'm in a hurry.
I'm currently struggling with my code to perform SFOCAL1 command. /DRDY goes down for the right amount of time, but nothing is written in OFCAL[2..0] registers. I came online to find some sample code could help me to point out where is the problem. May be there is something written elsewhere in datasheet I missed :-)

brendanmatkin, 2023-10-30:
I'll do some pull requests soon but I think this is an improved SFOCAL1 function.

_NOTE! I've made a couple of library changes that are used here_:
1.  startADC1 is modified, I've moved the pin selection from `readADC` to `startADC` as a fix to issue #5. I'll get that up here somewhere soon too.
2. I've added `drdy_pin` functions that duplicate the start pin functions (except with `_ads126x_setup_input` of course)

```cpp
void calibrateSelfOffsetADC1(uint16_t waitTime = 0);

// assumes that you want to calibrate with the same GAIN and VREF that you are using
// if not using DRDY pin, see table 9-28 on pg 79 to find appropriate wait time
void ADS126X::calibrateSelfOffsetADC1(uint16_t waitTime) {
  const uint8_t previousRunMode = REGISTER.MODE0.bit.RUNMODE;
  setContinuousMode();
  startADC1(ADS126X_FLOAT, ADS126X_FLOAT);
  _ads126x_delay(10); // let things settle, probably not necessary but 10ms for a calibration doesn't matter.. 
  ADS126X::sendCommand(ADS126X_SFOCAL1);
  if (drdy_used) {
    while (_ads126x_read_pin(drdy_pin) == HIGH) {
      // wait for calibration. drdy_pin is pulled low when cal is complete
    }
  } else {
    _ads126x_delay(waitTime);
  }
  // reset RUNMODE
  if (previousRunMode != ADS126X_CONV_CONT) {
    REGISTER.MODE0.bit.RUNMODE = previousRunMode;
    ADS126X::writeRegister(ADS126X_MODE0);
  }
}
```